adamws/kicad-kbplacer #2: Can't get plugin to work

malcx95, 2023-02-19:
Hi,

I'm trying to use the plugin but i can't get it to work, it just shows the following stack trace:

[image]

I've tried it both on Windows (Kicad 7.0) and on Linux (Kicad 7 and Kicad 6).
This happens on both the example files and my own json files converted with https://adamws.github.io/kle-serial/ as the instructions say. Do you know what I might be doing wrong?

adamws, 2023-02-19:
> This happens on both the example files 

please clarify which examples are you using. Did you try with [demo](https://github.com/adamws/kicad-kbplacer/tree/master/demo) project?

Also, version v0.2 (latest release) is not KiCad 7 compatible.

malcx95, 2023-02-19:
I just realized I missed the first part of the instruction, which is to create the matrix schematic. I also only used the layout json from the examples and not the whole project.

So I now installed Kicad 6.0 on Linux and downloaded the demo project in the demo folder. I open the demo project, go to the PCB Editor, but now when I select Tools > External plugins > Keyboard placer I am immediately met with this:

[image]

I have also as the instructions said copied the contents of `kbdplacer` to what Kicad says is the plugin directory, but it does not seem to make a difference.

malcx95, 2023-02-19:
Just tested with my computer and for some reason it works fine there. The both run Linux mint, but this time I installed kicad 6 directly without installing kicad 7 first. Maybe Kicad 7 doesn't completely uninstall properly? Anyway, it works for me now, thanks!

adamws, 2023-02-19:
I started work on KiCad 7 compatibility, should be available soon.

closed 2023-02-19 by adamws

sakgoyal, 2023-08-17:
is this updated for kicad 7 yet? its not working for me (windows 10). When I ran that python code to get the traceback, the result was empty. not sure what I should do now

adamws, 2023-08-17:
yes, KiCad 7 is supported since release v0.3
- what is exact version of kicad you are using (7.0.7 was released few days ago, it wasn't tested on that yet)
- `import pcbnew; pcbnew.GetWizardsBackTrace()` returns empty string if there are no errors while loading plugin. Check if it is listed in `Preferences->PCB Editor->Action Plugins`