adamws/kicad-kbplacer #3: Place additional components per-switch

cgarcia2097, 2023-03-13:
[My older codebase](https://github.com/cgarcia2097/kle-placement-router), that has fallen in disrepair, is able to place components other than diodes around the switch. Would you be able to implement that feature in this plugin?

adamws, 2023-03-14:
you can do it already. Run plugin twice, for second run place first component next to switch, check `Use first switch-diode pair as reference for relative position` and modify `Diode annotation format string` to match new components. This way your other components will be handled just like diodes.
Dedicated interface for this (which would work in single plugin run) might be added in some future release but it is not on top of my priority list.

cgarcia2097, 2023-03-14:
Understood. That being said, are you accepting pull requests? I could try patching it in.

ReFil, 2023-05-31:
This would be a really handy feature for placing per switch LEDs alongside switch and diode

adamws, 2023-05-31:
> This would be a really handy feature for placing per switch LEDs alongside switch and diode

This feature will be part of next release and it currently developed on https://github.com/adamws/kicad-kbplacer/tree/place-additional-elements branch.

closed 2023-06-20 by adamws