adamws/kicad-kbplacer #5: Exception on python  action plugin code

Pimentoso, 2023-06-16:
Hi,

trying to run the keyboard placer plugin but I'm getting this error.
[image]

Here is the keyboard json, converted from KLE using one of the tools mentioned in the guide.
https://pastebin.com/s0VmUtqL

Kicad info 
```
Application: KiCad PCB Editor arm64 on arm64

Version: 7.0.5-0, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 5.3.1
	FontConfig 2.14.0
	libcurl/7.88.1 (SecureTransport) LibreSSL/3.3.6 zlib/1.2.11 nghttp2/1.51.0

Platform: macOS Ventura Version 13.4 (Build 22F66), 64 bit, Little endian, wxMac

Build Info:
	Date: May 26 2023 08:14:40
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.80.0
	OCC: 7.6.3
	Curl: 7.77.0
	ngspice: 38
	Compiler: Clang 13.0.0 with C++ ABI 1002

Build settings:
	KICAD_SPICE=ON
```

adamws, 2023-06-17:
does it work when `Add Tracks` options unchecked?
which switch footprints do you use? can you share `.kicad_pcb` file and `kbplacer.log` file?

Pimentoso, 2023-06-20:
Thanks @adamws by looking at the kbplacer.log
```
22:04:27 com_github_adamws_kicad-kbplacer.kbplacer_plugin_action 35: Searching for SW1 footprint
```
I'm afraid it's looking for a symbol named SW1 while in my schema it's MX1. See attached. I probably have to rename my symbols

Here are the files you requested
https://files.catbox.moe/7s5wuq.kicad_pcb
https://files.catbox.moe/k3uop5.log

adamws, 2023-06-20:
No need to rename footprints, you can change default switch annotation from `SW{}` to `MX{}`
[image]
I've run plugin with yours board and it looks like number of switches does not match layout file - `Cannot find footprint MX54`
[image]
This will be the result: 
[image]
can you share initial json layout file (not internal form)?

Pimentoso, 2023-06-20:
Thanks @adamws for the hint, I can take it from here and do some tests. The error is probably because the board has a rotary encoder which has a diode, and the Dxx/MXxx naming is not consistent for the last few switches (the arrows at the bottom right)
I will let you know. Thanks again

closed 2023-06-20 by adamws