adamws/kicad-kbplacer #21: KeyPlacement not Following Column order

PabloOyarzo, 2024-02-02:
**Describe the bug**
Tried different layout export from Keyboard Layout Editor and it doesn't follow the column order 

**Additional context**
Heres kbplacer.log
[kbplacer.log](https://github.com/adamws/kicad-kbplacer/files/14143190/kbplacer.log)

**Screenshots**
[image]
[image]
[image]
[image]

**KiCad version info**
```
Application: KiCad PCB Editor x64 on x64

Version: 7.0.7, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 10 (build 19045), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Aug 14 2023 02:42:39
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.7.1
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

```

adamws, 2024-02-02:
`kblacer` places switches in keyboard-layout-editor order which is topmost leftmost, see example:
[image]
which is bit counterintuitive for ergo layouts.

I don't have clean solution for that yet.
1) for via-annotated layouts I could sort by labels
2) I could add option to sort by net-list (but that wouldn't work in all cases, in your example you would need to change column orders)

anyway, none of the above are there yet and for now the easiest fix is to rename switches on schematic.

adamws, 2024-02-02:
you could also modify schematic a bit so the switches placement mimics the physical placement and re-run annotate:
[image]

PabloOyarzo, 2024-02-02:
> alternatively, swap (select two and shift+S) switches on PCB and re-run plugin without layout file and diode placement enabled

This would do the trick for now thanks. 

---

Important to point out that with https://keyboard-tools.xyz/ with the following settings can get  the closest to the result I was looking for. But I had to change the name prefix afterwards. 

[image]
[image]

closed 2024-02-05 by adamws