adamws/kicad-kbplacer #22: Placement of Hot Swap Sockets on the Back Side

ij0n, 2024-02-09:
Hi,

Thanks for creating this awesome plugin! 

I'd love to place not normal keys, but Hot Swap Sockets, which would go on the backside of the PCB. The LEDs and Diodes need to be on the backside too. I have placed everything on the frontside, but when i flip the entire keyboard, it is mirrored. 

What I would love, if there could be a feature to select if the keys go on the front or backside, instead of placing them on the frontside by default.

Cheers
Best
ijon

adamws, 2024-02-09:
Hello, this is a good idea and it should be pretty easy to implement. I will include it in next release.

adamws, 2024-02-09:
btw, `kbplacer` does not modify side of switch footprints, if you flip switches before running plugin they will remain on  backside.

ij0n, 2024-02-10:
Hi Adam,

this is awesome - not only that you read and responded, but that you even like the idea. 

The proposed workaround totally works, love it! 

Best
ijon

closed 2024-02-12 by adamws