adamws/kicad-kbplacer #28: How can i place WS2812B LED's?

SnappierSoap318, 2024-04-20:
How can i add WS2812b's which have a diode annotation?

adamws, 2024-04-20:
easiest option:
use different annotation for LEDs
[image]
[image]

If both key matrix diodes and LEDs use same annotation there is an workaround (make backup first):
- clear annotations
- select LEDs on schematic, run annotation on selection - we want matching numbers between LEDs and switches
- annotate remaining symbols, use 'keep existing annotations'
you should have something like this:
[image]
- update PCB
- run plugin in two stages:
  - first: (note that diode placement option is unticked)
    [image]
    it should result in something like this:
    [image]
  - second (note that layout file is empty - we do not want to move switches anymore, also additional elements list is empty): 
    [image]
    should result in something like this:
    [image]
- go back to schematic, clear annotations, re-annotate (or revert to previous saved state)
  - use scope and numbering settings in annotation window
- update PCB
  [image]

SnappierSoap318, 2024-04-20:
Thanks for the detailed explanation, I did it the extra hard way, i off-setted the numbers variable in `place_switch_elements` in `key_placer.py` which worked much better than i expected

closed 2024-04-20 by adamws