adamws/kicad-kbplacer #42: StopIteration when attempting to place layout

kwesou, 2024-11-03:
**Describe the bug**
When I try to place my layout i get a StopIteration error. It returns the same with any mix of settings in the menu.

**Additional context**
`kbplacer.log` is in the zip file
The `firstattempt.json` and `secondattempt.json` are both JSON's I've attempted to use to get this to place, both unsuccessfully.  The log file is for the second.
[Right.zip](https://github.com/user-attachments/files/17611048/Right.zip)

**Screenshots**
[image]
[image]

**KiCad version info**
[image]

adamws, 2024-11-03:
appears that there are two problems here:
1) you are missing row/column labels in schematic
2) schematic does not match kle layout file - for example in the layout file first column (`6`) has three keys: `2,6`,`3,6` and `4,6` - on the schematic first column from left has 6 keys

[image]

kwesou, 2024-11-03:
I'm aware of both of those. I just made a matrix with enough components to see if it would place the keys correctly. Fixing both of those issues still doesn't let the addon place the keys properly.

adamws, 2024-11-03:
perhaps it does not work if matrix does not start with position (0, 0) - I can verify that later.
If this is a case its a bug and only solution would be to redo annotations

kwesou, 2024-11-04:
I have found the issue. This wasn't an issue with the addon but with my JSON files. No idea why the first one didn't work, but the second had the keys moved to the left incorrectly giving some of them impossible coordinates. I've included the now working layout as reference. Thanks for your help.
[keyboard-layout(1).json](https://github.com/user-attachments/files/17622663/keyboard-layout.1.json)

closed 2024-11-04 by kwesou