adamws/kicad-kbplacer #47: Cannot use the plugin because of a TypeError

Momossim, 2025-02-17:
**Describe the bug**
When i try to use the plugin (with everything default), I get the TypeError : setdiscard() takes exactly one argument (2 given). When i remove the switch annotation i don't get the error, but it obviously fails. I thought that i did something wrong, so i tried to export my librairies and my project on linux (lubuntu) on a vm with a clean installation of kicad, but i get the same error with the default settings. 

**Additional context**
I didn't find a kbplacer.log file.
kle layout : 
[eufmax.json](https://github.com/user-attachments/files/18829746/eufmax.json)
Apparently ".kicad_pcb" files are not allowed so i can't share it.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

[image]
error text : "Traceback (most recent call last):
  File "C:\Users\Momos\Documents\KiCad\8.0\3rdparty\plugins\com_github_adamws_kicad-kbplacer\kbplacer_plugin_action.py", line 92, in Run
    self.__run()
  File "C:\Users\Momos\Documents\KiCad\8.0\3rdparty\plugins\com_github_adamws_kicad-kbplacer\kbplacer_plugin_action.py", line 76, in __run
    run_from_gui(self.board_path, gui_state)
  File "C:\Users\Momos\Documents\KiCad\8.0\3rdparty\plugins\com_github_adamws_kicad-kbplacer\kbplacer_plugin.py", line 93, in run_from_gui
    return run(settings)
           ^^^^^^^^^^^^^
  File "C:\Users\Momos\Documents\KiCad\8.0\3rdparty\plugins\com_github_adamws_kicad-kbplacer\kbplacer_plugin.py", line 47, in run
    placer.run(
  File "C:\Users\Momos\Documents\KiCad\8.0\3rdparty\plugins\com_github_adamws_kicad-kbplacer\key_placer.py", line 941, in run
    key_matrix = KeyMatrix(
                 ^^^^^^^^^^
  File "C:\Users\Momos\Documents\KiCad\8.0\3rdparty\plugins\com_github_adamws_kicad-kbplacer\key_placer.py", line 107, in __init__
    switches_nets[switch_reference].discard(*common_nets)
TypeError: set.discard() takes exactly one argument (2 given)
"

[image]

[image]
**KiCad version info**
8.0
Use Help->About KiCad->Copy Version Info and paste it here
Windows : "
Application: KiCad x64 on x64

Version: 8.0.7, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.0.1
	FontConfig 2.15.0
	libcurl/8.10.1-DEV Schannel zlib/1.3.1

Platform: Windows 11 (version 22631), édition 64-bit, 64 bit, Little endian, wxMSW
OpenGL: NVIDIA Corporation, NVIDIA GeForce RTX 3060/PCIe/SSE2, 4.6.0 NVIDIA 572.42

Build Info:
	Date: Dec  3 2024 15:58:01
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.86.0
	OCC: 7.8.1
	Curl: 8.10.1-DEV
	ngspice: 43
	Compiler: Visual C++ 1939 without C++ ABI

Build settings:
"
Linux : "Application: KiCad x86_64 on x86_64

Version: 8.0.8-8.0.8-0~ubuntu24.04.1, release build

Libraries:
	wxWidgets 3.2.4
	FreeType 2.13.2
	HarfBuzz 8.3.0
	FontConfig 2.15.0
	libcurl/8.5.0 OpenSSL/3.0.13 zlib/1.3 brotli/1.1.0 zstd/1.5.5 libidn2/2.3.7 libpsl/0.21.2 (+libidn2/2.3.7) libssh/0.10.6/openssl/zlib nghttp2/1.59.0 librtmp/2.3 OpenLDAP/2.6.7

Platform: Ubuntu 24.04.2 LTS, 64 bit, Little endian, wxGTK, X11, LXQt, x11
OpenGL: VMware, Inc., SVGA3D; build: RELEASE;  LLVM;, 4.1 (Compatibility Profile) Mesa 24.2.8-1ubuntu1~24.04.1

Build Info:
	Date: Jan 14 2025 02:26:31
	wxWidgets: 3.2.4 (wchar_t,wx containers) GTK+ 3.24
	Boost: 1.83.0
	OCC: 7.6.3
	Curl: 8.5.0
	ngspice: 43
	Compiler: GCC 13.3.0 with C++ ABI 1018

Build settings:

"

adamws, 2025-02-17:
>  Apparently ".kicad_pcb" files are not allowed so i can't share it.

you can zip it and then attach

Momossim, 2025-02-17:
> > Apparently ".kicad_pcb" files are not allowed so i can't share it.
> 
> you can zip it and then attach

oh i didn't know that worked. i'll put it here and in the original post

[keyboard pcb.zip](https://github.com/user-attachments/files/18830682/keyboard.pcb.zip)

adamws, 2025-02-17:
please attach schematic as well, it seems odd that some diodes have two ends connected to switch (for example SW1-D3)

adamws, 2025-02-18:
I think here is your mistake:

[image]

that's why ROW0 is missing on PCB file (resulting in first row diodes to be connected to switches incorrectly)

Momossim, 2025-02-18:
Thank you it worked. But why does it say that there is multiple diodes per switch layouts when i try to use the default diode preset ?

[image]

adamws, 2025-02-18:
each `SW{}` footprint must have exactly one `D{}` footprint connected in order for 'Default' to work. Connections are found based on nets.

Momossim, 2025-02-18:
Thank you. I had two mislabelled switches, but I just checked every symbols with the SW{} notation and they all have a diode. btw the Relative position option do that : 

[image]

adamws, 2025-02-18:
which may be correct - depends what was the starting state. Relative position duplicates diode placement of first switch-diode pair.

Momossim, 2025-02-18:
I switched the position of the diode (it was on the right side, i put it on the left) on the schematic and it worked. Thank you.

closed 2025-02-18 by Momossim